nauful/LibUA #129: Erro monitoring

franklupo, 2023-07-26:
Hi, 
i connect client to Siemens plc. 
When read the statuscode is null but value and timestamps are ok. 
When create monitoring receive status **BadEncodingLimits Exceeded**

what am I doing wrong?

Best regards

nauful, 2023-07-26:
Hello,

Can you share a Wireshark capture with no security?

franklupo, 2023-07-26:
i don't know how to do with Wireshark

nauful, 2023-07-26:
https://help.salesforce.com/s/articleView?id=000385344&language=en_US&type=1

Otherwise you might be monitoring too many items in a single call. Try batches of a few items at once.

franklupo, 2023-07-27:
[opcua.zip](https://github.com/nauful/LibUA/files/12180724/opcua.zip)

franklupo, 2023-07-27:
Hi,
I don't know if it helps but with this library it works 
https://github.com/convertersystems/opc-ua-client

best regards

nauful, 2023-07-27:
Looks like a `PublishResponse` is too large for the sender/receiver according to the initial configuration. I need to see a capture with no security, this connection is secured with encryption so I cannot read the messages.

franklupo, 2023-07-27:
[opcua2.zip](https://github.com/nauful/LibUA/files/12184549/opcua2.zip)

closed 2023-07-27 by nauful

franklupo, 2023-07-27:
encryption where should I disable?

nauful, 2023-07-27:
https://github.com/nauful/LibUA/blob/7e1445072a418bcb81fbe98a9ab67452147c4dc6/NET%20Core/TestClient/Program.cs#L120

Change `SignAndEncrypt` to `Sign`.

franklupo, 2023-07-27:
```
var messageSecurityMode = MessageSecurityMode.None;
var securityPolicy = SecurityPolicy.None;
```

franklupo, 2023-07-27:
i'm using the port 4841 and the wireshark is setting for 4840.
try now

[opcua3.zip](https://github.com/nauful/LibUA/files/12185418/opcua3.zip)

franklupo, 2023-07-27:
Reopen issue please

nauful, 2023-07-27:
I can try to assist you with library usage but I prefer to keep issues for problems with the library itself. Please try my suggestions above and let me know if that works.

franklupo, 2023-07-27:
The problem is the parameter filter in **MonitoringParameters**,
if present response **Goof** is null return **BadEncodingLimitsExceeded**
IF presente the notification not work.

nauful, 2023-07-27:
Thanks, will note that if I see this issue again.