nauful/LibUA #132: Unable to pass "null" as filter in MonitoredItemCreateRequest

valerionatangelo, 2023-09-06:
Hi,

passing `null` as filter in the `MonitoringParameters` makes the `CreateMonitoredItems` call fail (BadEncodingLimitsExceeded).
This seems to happen because the implementation of the new method

```
Encode(this MemoryBuffer mem, MonitoringFilter filter, bool includeType)
```

At the same time, the TestClient project still passes `null` for the filter parameter.
Please note that with the old Nuget 1.0.8 I was able to pass `null` as filter.

If I try to pass a filter constructed for example like the following:

```
new DataChangeFilter(DataChangeTrigger.Status, DeadbandType.None, 0);
```

then I get a BadEndOfStream error.

It's not clear to me what to pass in the filter argument to make the call work.
Thanks.

nauful, 2023-09-07:
You're correct, `null` should still work. Updated master, please try the TestClient and TestServer applications again.

valerionatangelo, 2023-09-08:
I can confirm the problem is gone.
Thanks.

closed 2023-09-08 by nauful