nauful/LibUA #133: Problem with freeOPAUA

thomasschichl, 2023-10-30:
Hi, first of all thank you for providing this great framework.
I am using you libraries in unity for educational issues and everything is working pretty smoothly. However, we are facing one problem. As we are using rasperry Pi systems and python as a programming language as well, we are using freeopcua
https://github.com/FreeOpcUa/python-opcua

When we create a simply server in python (on windows or Pi systems) and try using your library, we cannot connect to establish a communication with the server. It is possible to connect to the server. However it is not possible to open a secure channel or start the session...
For each Method (opensecurechannel, createsession, activatesession) I get a BadEncodingLimitsExceeded message. Using other OPC UA servers everythin is just working fine (different PLC OPC UA Servers like Siemens, Codesys, B&R,...)
I tried creating my own Client in unity and I tried your TestClient directly in Visual Studio. Always BadEncodingLimitsExceeded Errors.

I am using the server_minimal from the freeopcua examples right now. Every other client is able to connect to the server- only the LibUA not...

I have already worked several hours and did a lot of researche rearding the error message- however, no luck. Do you have an idea on how to solve this problem?

nauful, 2023-10-30:
Hello Thomas, sorry to hear that you have such a strange issue.

If I understand correctly, you're using LibUA as the client to connect to a server? If so, could you please try connecting with None for security and give me a wireshark capture of the connection attempt?

thomasschichl, 2023-10-30:
Hi, thank you for the qick reply.  Yes LibUA is the client. However, I was just able to solve the "problem". Actually it was me not reading the updated repo [of python/opcua](https://github.com/FreeOpcUa/python-opcua) - the library is deprecated. According to description it should be usable with python >3.4. 
However they provide a new framework for python client/ server and with this framework everything works just fine and I can open the com with yout LibUA client.
It would, of course be great to know why the communication is rejected and I can provide a wireshark screenshot- however, as the python-opcua is not supported any longer I am not quit sure if its worth the effort- just let me know.

nauful, 2023-10-30:
Yes, I am definitely curious why this was rejected. I would like a Wireshark capture of the initial connection process.

Otherwise glad to hear your problem was easily resolved. Please do feel free to open an issue here before you spend much time diagnosing, and let me know if I can assist your project in the future. I would say 95% of issues I've experienced/seen have been solved by looking at what's missing from a Wireshark capture, usually quick.

thomasschichl, 2023-10-30:
[image]

thomasschichl, 2023-10-30:
Here you are- no security on the server...

thomasschichl, 2023-10-30:
I would assume the [Calculated window size: 2161152] is the issue...

nauful, 2023-10-30:
Can you drag and drop the capture here? I don't see an attachment.

thomasschichl, 2023-10-30:
[image]

thomasschichl, 2023-10-30:
> Can you drag and drop the capture here? I don't see an attachment.

[image]

closed 2023-10-30 by nauful

thomasschichl, 2023-10-30:
> I mean the wireshark capture file, not the screenshot. But if it's working now, at least there's no pending problem.

[opc.zip](https://github.com/nauful/LibUA/files/13209878/opc.zip)

nauful, 2023-10-30:
Thanks, I'll work with that. Do let me know if you run into any other issues.

thomasschichl, 2023-10-30:
> Thanks, I'll work with that. Do let me know if you run into any other issues.

Thanks a lot- and again, great framework- we are using it at our university and really appreciate the effort!

nauful, 2023-10-30:
Out of curiosity, which university uses this library?

thomasschichl, 2023-10-30:
> Out of curiosity, which university uses this library?

I am working at the university of applied science Upper Austria- the university is in Wels, Austria. In our excercises for digital factory, we create digital twins in Unity3D. We use industrial components like robot systems, PLC systems but also Windows and Pi systems as MQTT broker or OPC UA Servers. The Applications we create run on Windows and Android plattforms (Smartphones, Meta Quest Goggles, Neo Pico Goggles, MS Hololens 2,...).
For several terms we were using the demo license from OPCUAFX bei traeger. However, your LibUA framework is the only one we found so far which was executable on the MS Hololens 2. This was the reason why we switch to your framework.
Of course we always mention you and you github repo...

nauful, 2023-10-30:
Very cool, thanks for letting me know!

Do mention me if possible, and feel free to get in touch whenever I can offer help: nauful@quantensystems.com

thomasschichl, 2023-10-30:
> Very cool, thanks for letting me know!
> 
> Do mention me if possible, and feel free to get in touch whenever I can offer help: [nauful@quantensystems.com](mailto:nauful@quantensystems.com)

I just sent a request to my supervisor, that we add you website quantensystems.com. I assume, that it is ok. So I will send you the link as soon as I have it...

And just for your information- realvirtual.io is an unit yasset which provides features for virtual commissioning. They were using opc ua fx by traeger as well. When we told them that your framework is working on the Hololens, they switched to your framework. http://realvirtual.io/en/
So we try to be as open as possbile and share our sources not only with our students but also with company- I hope that is ok for you...

So thanks again and I will send you the link asap

nauful, 2023-10-30:
Of course, my library's licensing terms are OSS and free for commercial use. Glad it's been of use!