nauful/LibUA #134: Can not write value to my variable

petruh1n, 2023-11-01:
Good afternoon!
I started your server and added my variable to the tree.
[image]

I connect to the server using a third-party client (Prosys client).
I get an error when recording.
[image]

Please tell me what I'm doing wrong?

nauful, 2023-11-01:
Hello,

Nodes on the server are not writable by default. If you want to write to a node, you must handle storing that value for later reads.

Implement this virtual method in your `Application`:
https://github.com/nauful/LibUA/blob/837d0072a338e690d1c30ec32649499ae12ee10c/NET%20Core/LibUA/Application.cs#L474C4-L474C88
The default implementation returns `BadNotWritable` for write attempts. If your node is writable, you can store the new value (to return on reads, possibly also store in a historical time series) and return `Good` for that node. If your node isn't writable, you can return `BadNotWritable`.

`WriteValue` is determined here:
https://github.com/nauful/LibUA/blob/837d0072a338e690d1c30ec32649499ae12ee10c/NET%20Core/LibUA/Types.cs#L5876
This includes `NodeId` and `Value`.

nauful, 2023-11-01:
To return the written value when a node is read, look at this:
https://github.com/nauful/LibUA/blob/837d0072a338e690d1c30ec32649499ae12ee10c/NET%20Core/TestServer/Program.cs#L276

And this is to handle historical reads:
https://github.com/nauful/LibUA/blob/837d0072a338e690d1c30ec32649499ae12ee10c/NET%20Core/TestServer/Program.cs#L305

petruh1n, 2023-11-02:
I return Good in the HandleWriteRequest method, the error no longer appears, but the value is still not displayed. 
I didn't quite understand what it means to store the new value. 
Could you share some example, please?

petruh1n, 2023-11-02:
I managed to write down the value in this way. Is this the right decision?
[image]

nauful, 2023-11-03:
Look at this sample code:
```
			private Dictionary<NodeId, DataValue> writtenValues = new Dictionary<NodeId, DataValue>();

			public override uint[] HandleWriteRequest(object session, WriteValue[] writeValues)
			{
				var respStatus = new UInt32[writeValues.Length];
				for (int i = 0; i < writeValues.Length; i++)
				{
					if (writeValues[i].AttributeId == NodeAttribute.Value)
					{
						writtenValues[writeValues[i].NodeId] = writeValues[i].Value;
						MonitorNotifyDataChange(writeValues[i].NodeId, writeValues[i].Value);
						respStatus[i] = (UInt32)StatusCode.Good;
					}
					else
					{
						respStatus[i] = (UInt32)StatusCode.BadNodeAttributesInvalid;
					}
				}

				return respStatus;
			}

			protected override DataValue HandleReadRequestInternal(NodeId id)
			{
				if (writtenValues.TryGetValue(id, out DataValue value))
				{
					return value;
				}

				Node node = null;
				if (id.NamespaceIndex == 2 &&
					AddressSpaceTable.TryGetValue(id, out node))
				{
					if (node == Node1D)
					{
						return new DataValue(new float[] { 1.0f, 2.0f, 3.0f }, StatusCode.Good, DateTime.Now);
					}
					else if (node == Node2D)
					{
						return new DataValue(new float[2, 2]
						{
							{ 1.0f, 2.0f },
							{ 3.0f, 4.0f }
						}, StatusCode.Good, DateTime.Now);
					}
					else
					{
						return new DataValue(3.14159265, StatusCode.Good, DateTime.Now);
					}
				}

				return base.HandleReadRequestInternal(id);
			}
```
			
You need to store the written value somewhere, publish data change, then return it on read.

petruh1n, 2023-11-03:
Thank you very much.
Why do I need to return a value in HandleReadRequestInternal?

nauful, 2023-11-03:
For read internal, you need to return the value of the node involved. This is the sent back to the client.

closed 2023-11-07 by petruh1n