nauful/LibUA #139: UaClient error to read a matrix

VitorKawao, 2023-12-07:
Hello Nauful,

I am having some problems when I try to read a matrix (array with two dimensions).
With an array with 1 dimension, it works perfectly

I am using your api to connect to KepServer.
I create there a matrix 3x3

[image]

Using Kepware Click Client, it can read correctly: [0;0;0][0;0;0][0;0;0]

[image]

When I try to read with you api, it returns an ushort[9], but with a invalid 2

[image]

But if I change the values of the matrix, it returns null

[image]
[image]

Am I doing something wrong?

Thank you,
Vítor Guedes

nauful, 2023-12-07:
Hello Vitor!

Could you please rerun that test while capturing with Wireshark and security mode set to None? I wonder what data is coming back. I may be deserializing incorrectly, 2D matrix has been a rare case for testing. I can look at the Wireshark capture to see what was read.

Thanks,
Nauful

VitorKawao, 2023-12-07:
I downloaded the latest version, but it still has the same errors

nauful, 2023-12-07:
Hello Vitor,

Do you have this commit?
https://github.com/nauful/LibUA/commit/cef29f87ec0c2b2b44c8077b065390de97721897

I am able to reproduce the issue before but not after this change. Let me know if that still fails with the latest commit.

Thanks,
Nauful

VitorKawao, 2023-12-07:
I am sorry, I think I haven't.
Now I have it, and it worked

[image]

Thank you,
Vítor Guedes

nauful, 2023-12-07:
Hello Vitor,

Glad it worked! Feel free to let me know if you come across anything else.

Happy holidays,
Nauful

closed 2023-12-07 by nauful