nauful/LibUA #140: How to read an ExtensionObject?

StarlimsDanielD, 2023-12-20:
I used the LibUA.Client.Read() functionality to read a value from a specific node and this value contains an ExtensionObject.
It's contents are something like this
```
{
	"TypeId": {
		"NamespaceIndex": 0,
		"NumericIdentifier": 886,
		"ByteStringIdentifier": null,
		"StringIdentifier": null,
		"IdType": 0
	},
	"Body": "AAAAAAAAAAAAAAAAAABZQA==",
	"Payload": null
}
```
The way I understand this is that I need to somehow get the structure of the object definition used for this in order to read it's contents. I do see something about the type, but I have no clue how to use it. I also looked into the body part and it contains a byte array which does not make sense at this point.

The value that is sent here is of the type RANGE and contains a low and high value (0-100). Is this supported by the LibUA and if so, could you please point me in the right direction on how to read this?

Thanks!

nauful, 2023-12-21:
Is the content a string of json or a binary encoded blob?

Look at the MemoryBuffer class and its decoding methods to read from ExtensionObjects. If it's a string, you can decide a string. If it's a complex type, you can decide the fields in order using decode methods with the appropriate variable types.

StarlimsDanielD, 2023-12-22:
In my particular case, it is a byte array that should represent a date time, but it is a custom complex type. 

> you can decide the fields in order using decode methods with the appropriate variable types
This is the part I'm confused on, do you have a small example on how to do this? This is the part I am missing, it seems.

nauful, 2023-12-22:
Here's an example of how to decide some fields in order from a memory buffer. Load a memory buffer with the data (byte array) from an extension object.

https://github.com/nauful/LibUA/blob/d02a2b00228135eeefbf5203bdb988a879467baa/NET%20Core/LibUA/Client.cs#L1923

closed 2023-12-22 by nauful

nauful, 2023-12-22:
MemoryBufferExtensions.cs has more examples. Look at Decode methods near the end of that file for complex structures.

StarlimsDanielD, 2024-01-24:
I finally had time to look at this again, but I'm not sur we're on the same page.
The code I have actually already uses the MemoryBuffer and in this case it is to build a DataValue;

 `public static bool Encode(this MemoryBuffer mem, DataValue dv)`

The DataValue that is build has a value that seems to contain an ExtensionObject:

`{
	"TypeId": {
		"NamespaceIndex": 0,
		"NumericIdentifier": 889,
		"ByteStringIdentifier": null,
		"StringIdentifier": null,
		"IdType": 0
	},
	"Body": "LwAAAGh0dHA6Ly93d3cub3BjZm91bmRhdGlvbi5vcmcvVUEvdW5pdHMvdW4vY2VmYWN0SEFGAAIDAAAAwrBGAhEAAABkZWdyZWUgRmFocmVuaGVpdA==",
	"Payload": null
}`

Now, inside this ExtensionObject seems to be a byteArray inside the body. According to the UA Expert client software it is an object with multiple properties, but my question is how do I get that object with those properties?

nauful, 2024-01-25:
It's possible to have a MemoryBuffer work on an arbitrary byte buffer then copy that to/from a DataValue's value.

If your body is Base64 encoded, you can use the standard C# libraries to convert that to a byteArray. You can then create a new MemoryBuffer passing byteArray as input, then use the various extension methods to decode.

If you're trying to build that object, create a new MemoryBuffer with a large byte array, use various extension methods to encode fields, then take the slice of the array [0, MemoryBuffer.Position), Base64 encode, and set as the value.