nauful/LibUA #142: TranslateBrowsePath does not mark the final BrowsePathTarget correctly

fjscherp, 2024-02-02:
Hi,

Thanks for making this library, it helped me a lot. 
Some context: I'm using your library to make a server side OPC-UA simulator as alternative for the real thing:  a Beckhoff PC. And I talk to it using a already working OPC UA client based on a Unified Automation client library. To make all it work with LibUA I had to solve some issues:

* A bug (I think) in the HandleTranslateBrowsePathRequest routine. It relates to the RemaingPathIndex in BrowsePathTarget which should have a special value for the last element in the list (UInt32.MaxValue).  I couldn't find that in your library (1.01.14) so had to adjust that ot make it work with the client library.
* Complete your method call implementation including support for the 'Argument' structure through ExtensionObject. Needed that because method discovery is used.
 
I added a patch to this ticket with the changes I made. Could be useful for other people.
I don't know if this the correct way to solve it but I leave that up to you. 

I hope you keep maintaining this library. Thanks

[Extension_+_bugfix_to_support_OPC_methods.patch](https://github.com/nauful/LibUA/files/14138882/Extension_%2B_bugfix_to_support_OPC_methods.patch)

nauful, 2024-02-02:
Hello,

Glad you like the library and thanks for sharing. Patches to problems are always appreciated! I will take a look at that and merge your fixes in after some testing.

nauful, 2024-02-02:
Done: https://github.com/nauful/LibUA/commit/1b71cecd01771473063ccb9123a58ef11013a973

closed 2024-02-02 by nauful