nauful/LibUA #143: Exception in  MemoryBufferExtension.cs with a Prosys OPCUA Server

game4automation, 2024-02-20:
Hi Nauful,

there is a exception coming up when testing with the Prosys OPCUA Simulation Server. It is the throw new Exception in Line 1833, Decode in MemoryBufferExtension. It is a  unknown encoding Mask of 64.

We are browsing the nodes by this call:
```
 client.Browse(new BrowseDescription[]
            {
                new BrowseDescription(
                    topnodeid,
                    BrowseDirection.Forward,
                    NodeId.Zero,
                    true, 0x0000003Fu, BrowseResultMask.All)
            }, 10000, out browseResults);
```

Any Idea how to prevent this?

Thanks a lot 
Thomas (realvirtual.io)

game4automation, 2024-02-20:
Hi, where do I get the code change.

For Wireshark I am not an expert, how to do a fast capture (for beginners?)

Thanks
Thomas

nauful, 2024-02-20:
Either this repo, or you can update to the latest nuget package I pushed: https://www.nuget.org/packages/nauful-LibUA-core

Let's see if the new nuget package fixes your issue. If not, to capture in wireshark, click Capture -> Start on the menu, connect with security and encryption set to None, reproduce this error, Capture -> Stop on the menu, File -> Save As and upload the capture here. More info: https://www.wireshark.org/docs/wsug_html_chunked/ChCapCapturingSection.html

Thanks,
Nauful

game4automation, 2024-02-20:
[libuacapture.zip](https://github.com/nauful/LibUA/files/14350454/libuacapture.zip)

Hi, problem still exists even with the latest commit. Here is the wireshark capture. Thanks a lot.
Thomas

nauful, 2024-02-20:
I don't see the opcua capture in there, please check that you captured from the correct network adapter (or loopback adapter if localhost). Here's how to check: https://opcconnect.opcfoundation.org/2017/02/analyzing-opc-ua-communications-with-wireshark/

Do you still get an exception in the latest commit? If so, where in the code is that?

Thanks,
Nauful

nauful, 2024-02-20:
For the captures, can you tell me what port you connected on?

Also, there's this code change:
https://github.com/nauful/LibUA/commit/dddad72459f1064726908224b432f866ae57a3ce#diff-1ce579e93332054aa094abe3152192bcde0709fc1586b02271917b0d2d6a518f

Can you tell me the value of `encodingMask & 0x3F`? If `encodingMask` is 64, that should be 0, and goes to the TwoByte case.

Thanks,
Nauful

game4automation, 2024-02-20:
With your change now it throws an exception with encoding Mask 111

nauful, 2024-02-20:
I don't see any NodeId with a mask of 111/0x6F in the capture, perhaps a new capture?

Can you tell me where the parent is in the call stack? That will tell me what field it's trying to decode here.

game4automation, 2024-02-20:
Thanks a lot but again an exception at the same location but a different call stack:
[image]

game4automation, 2024-02-20:
You should be easily able to check this - we are testing with standard Prosys OPCUA Simulation Server:
[image]

nauful, 2024-02-20:
I don't think this is the same issue. I was able to successfully decode your wireshark capture after the previous fix.

This looks like opening a secure channel is failing, which isn't something I've touched recently. Does it work with security set to None?

game4automation, 2024-02-21:
You are right - it seemed to be to late yesterday. Now it works. Thanks a lot for your fast support.

nauful, 2024-02-21:
Fantastic!

closed 2024-02-21 by nauful