nauful/LibUA #148: EUInformation and Range

bjornfe, 2024-03-01:
Is it possible to implement the types for EUInformation and Range ?

nauful, 2024-03-01:
You can add the following methods to `MemoryBufferExtensions` for each of `Encode`, `Decode` and `CodingSize`. Otherwise I will add them to my TODO list for now.

closed 2024-03-01 by nauful

nauful, 2024-03-02:
Merged in your PR, good changes!

By the way, if you are using this library for any projects, I would be very interested in hearing where and how. I would appreciate your permission to use that information in my list of use cases. If you prefer email, my email is [nauful@quantensystems.com](mailto:nauful@quantensystems.com)

bjornfe, 2024-03-02:
Hi. Yes we are looking into using it for a commercial project. I do not have a ready to distribute information sheet yet for the project, but if you are interested we can have a Teams call and I can show you what we have?. Here is a link to our software: [](https://ixys.no/vju/)
We have not yet decided to use this library, but we are in testing phases where we are looking into replacing the Traeger implementation due to excessive memory usage and slow startup times.

nauful, 2024-03-02:
That makes sense. If it works and you choose this library, I would like to take a look then.

If you're looking at excessive memory usage and slow startup times, in `Application.cs`, `SetupDefaultAddressSpace` can be greatly reduced if your clients don't rely on that. Otherwise you can virtualize nodes within the application's address space without creating heavyweight objects e.g. `Dictionary<NodeId, ValueQualityType>` for reads/writes.

Let me know if I can assist with your project in any way.