nauful/LibUA #153: Browsing S7 OPCUA Server, Outputs not returned (FunctionalGroupType?)

game4automation, 2024-03-18:
Hi Nauful,

we are trying to browse Nodes in a S7 OPCUA Server. We are browsing with this command:
    client.Browse(new BrowseDescription[]
    {
        new BrowseDescription(
            topnodeid,
            BrowseDirection.Forward,
            NodeId.Zero,
            true, 0xFFFFFFFFu, BrowseResultMask.All)
    }, 10000, out browseResults);

In OPCWatch we are seeing several Outputs (Q_Test....) but the browse result by the code above just gives us 2 results, the icon and a Node FunctionalGroupType of NodeClass ObjectType. Do you know how to browse the Outputs?

Thanks a lot
Thomas

[image]
[image]

nauful, 2024-03-18:
Hello Thomas,

Could you share a Wireshark capture of browsing without encryption? I would guess it has something to do with references but I need to see to make sure.

Thanks,
Nauful

game4automation, 2024-03-18:
Hi Nauful,

here is the capture:
[opcuacapture.zip](https://github.com/nauful/LibUA/files/14639888/opcuacapture.zip)

nauful, 2024-03-18:
Can you capture the browse operation in the other client which shows the correct list of children?

Thanks,
Nauful

game4automation, 2024-03-18:
It should be somewhere in here:
[opcuacapture2.zip](https://github.com/nauful/LibUA/files/14640258/opcuacapture2.zip)

nauful, 2024-03-18:
Looks like Browse returns a continuation point for you to call BrowseNext:
```
client.Browse(new BrowseDescription[]
{
		new BrowseDescription(
			new NodeId(2, 0),
			BrowseDirection.Both,
			NodeId.Zero,
			true, 0xFFFFFFFFu, BrowseResultMask.All)
}, 1000, out browseResults);

while (browseResults[0].ContinuationPoint != null)
{
	client.BrowseNext(new[] { browseResults[0].ContinuationPoint }, false, out browseResults);
}
```

At every step, browseResults will be recreated as a new array.

game4automation, 2024-03-19:
Hi Nauful,

thanks a lot for your great support again. This resolved our issue. Something I did not knew.

Best regards
Thomas

nauful, 2024-03-19:
Great! It's uncommon but if results overflow `Browse`, servers support continuation points to continue a browse with `BrowseNext`.

Thanks,
Nauful

closed 2024-03-19 by nauful