nauful/LibUA #159: Regarding client connection issues

yongzhi0113, 2024-05-09:
When using opcua client, connect to a specific Opcua Server machine.
Calling OpenSecureChannel() will succeed.
But calling FindServers() and GetEndpoints() will fail.

When the same Opcua Server is connected via UaExpert, the following message will appear.
If we press the agree button, we can successfully connect to Opcua Server.
I guess this may be the problem. How can I troubleshoot this problem?

===========================================
GetEndpoints failed using the discov overy URL opc.tcp://MX6PLATFORM0001:4840/ returned from FindServers. Do you want to try the URL opc.tcp://192.168. 1.15:4840/ with the hostname used to call FindServers?

The hostname of the discovery URL used to call GetEndpoints (mx6platform0001) was replaced by the hostname used to call FindServers (192.168.1.15) Do you also want to replace the hostnames of the EndpointURLs with this hostname?

nauful, 2024-05-10:
If you try to connect via hostname, you need to do this to resolve the IP and connect via IP: https://stackoverflow.com/questions/68636144/c-sharp-resolve-hostname-with-dns-to-ip-address/68646923#68646923

If that still doesn't work, record a Wireshark capture with security disabled of the case where the calls fail and I can take another look.

nauful, 2024-05-13:
Can you record a Wireshark capture with security disabled? I can look at that. Otherwise I don't have enough information to find a problem.

yongzhi0113, 2024-05-14:
Sorry, there is no way to provide Wireshark capture.
I will close this issue.

closed 2024-05-14 by yongzhi0113