nauful/LibUA #160: Chunk messages are not created.

mosseedwin, 2024-05-29:
The server does not create chunk messages when it exceeds the buffer size.

nauful, 2024-05-29:
That functionality is there. Can you provide a reproduction case and more information?

mosseedwin, 2024-05-29:
Modify TestServer project.

                TrendNodes = new NodeVariable[1000]; to TrendNodes = new NodeVariable[2000];

Using Matrikon OPC UA Explorer to browser to "Items" node, the Explorer generate "badtcpmessagetoolarge".

nauful, 2024-05-29:
Tested with 10k nodes and UAExpert correctly browsed. Check that your connection settings allow large messages, I suspect that might be the issue. Can you share your client's installer?

mosseedwin, 2024-05-30:
Installer
https://drive.google.com/file/d/1Ni9j-hBetbimtoWqgsDYAvamUd6Yj17B/view?usp=sharing

nauful, 2024-05-31:
Your link gives me a 404. Please verify the maximum message size for your client. If the sum of chunks exceeds that, you get the error seen above. Couldn't reproduce.

closed 2024-05-31 by nauful

mosseedwin, 2024-05-31:
> Your link gives me a 404. Please verify the maximum message size for your client. If the sum of chunks exceeds that, you get the error seen above. Couldn't reproduce.

The test server sent ReceiverBufferSize and SendBufferSize with value 65535, but generates a message of 480072 bytes at TCP level.

https://wetransfer.com/downloads/8ee900ff6ae592c861eedec61044899a20240531155644/0a4ee4216b1a1d81af0d02528d0d4e9720240531155700/c0579e?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

nauful, 2024-05-31:
The server's acknowledgement has a maximum message size of 1 MB, which is 17 64k chunks:
[image]

You can change that here:
```
var app = new DemoApplication();
var server = new LibUA.Server.Master(app, 7718, 10, 30, 100, new DemoLogger(), MaximumMessageSize: 8 << 20);
server.Start();
```

That should fix your issue. A better solution here is to use Browse for the first 100 nodes and BrowseNext with continuation points for every 100 nodes left. For example, uaexpert:
[image]

Your dump:
[image]