nauful/LibUA #167: TIMEOUT: get Server Status

jvanlangen, 2024-06-13:
When using this library (.NET Framework 4.8) causes the clients giving a TIMEOUT on the get Server Status.

Any thoughts?

nauful, 2024-06-14:
Can you try reading server status on the server using the TestServer project? That's currently working for me.

bevanweiss, 2024-06-17:
@jvanlangen any update on this?
What version of the library are you using?
You might also need to show some Wireshark logs of the 'conversation' showing the TIMEOUT error, and some information around what your application logic is that's using the library...  are you sure you're calling base.HandleReadRequestInternal(NodeId id) if the node isn't one of your application nodes?

jvanlangen, 2024-06-20:
A, I've found the issue. We override the `HandleReadRequestInternal` method but didn't call the `base.HandleReadRequestInternal`. It's fixed now..

closed 2024-06-20 by jvanlangen