nauful/LibUA #169: High CPU

jvanlangen, 2024-06-20:
At the moment we suffer from high cpu usage when we activate the LibUA server. Any known issues with that?

My plan:
- Cloning the repo.
- Put it as a project in our solution and update the reference.
- Try the find the high cpu thing...

jvanlangen, 2024-06-20:
I've found something that might be causing hi-cpu.

https://github.com/nauful/LibUA/blob/df0ff4252b8caebe372b259882e3f34f6e380a40/NET/LibUA/NetDispatcherBase.cs#L196-L199

The problem is, that when there is not data, it will directly continue the while loop. There is no sleep/wait event here.
I just added a sleep(1) which brings the cpu usage down dramatically.

Any thoughts?

bevanweiss, 2024-06-20:
This is my breakage, I didn't notice that it's caused high CPU.
What I should have probably done is do the ReadAsync() and a TimerAsync (driven by the NeedsPulse), await either.. check if the Read is complete, if it is, go and do the read stuff.. otherwise if the NeedsPulse is ready, go and do the timer stuff (then reset timer task.. or just set it as autoresetting) then go back to waiting for them..

But short term... you could just check out the previous revision 990a7ac514ff5ad4834e0cc00d68fc853d851c33

I'll try to fix this up in the next week or so.

nauful, 2024-06-20:
Reverted back to the previous version. Try the latest master and let us know if you still see high CPU usage.

closed 2024-06-20 by nauful

jvanlangen, 2024-06-21:
I also found some small points in the code, which are not quite nice in terms of code structure. For example `(stream as NetworkStream).DataAvailable`, now you can be sure that `stream` is a `NetworkStream`, but if this ever changes you will get a `NullReferenceException` here. NullReferenceExceptions are generally seen as sloppiness. It's better to do a hard cast here. `((NetworkStream)stream).DataAvailable`, if the stream is not `NetworkStream`, you will get a `InvalidCastException` which explains the error better.

Would you mind if I make some pull requests?

nauful, 2024-06-21:
Sure, feel free.