nauful/LibUA #171: How to subscribe to an ExtensionObject or nested value array

Christ0ph89, 2024-07-01:
Hi Nauful,

I'm trying to subscribe to an ExtensionObject, which contains two sub-levels of value arrays, without success actually. I'm note sure, if it is even possible, since in the opc ua specification is written:  _Objects and views can be used to monitor Events_ [Ref](https://reference.opcfoundation.org/Core/Part4/v104/docs/5.12.1). 

Why I'm asking, if I subscribe to that specific node id in UAExpert from UnifiedAutomation, the Source and Server Timestamp is updated all the time on any value change in the sub tree of that node. If I subscribe, with NodeAttribute.Value, to that specific node in LibUA, nothing happens. Also UAExperts tell me, it's an mentioned ExtensionObject.

Can you tell me what I'm missing or is it a feature that is still not implemented in LibUA? 

Here is an picture of the attributes of that specific node, which shows the sub-levels of the value arrays:
[image]

Magniveo, 2024-07-02:
Hello, you can see the implementation in the TestClient project, and the implementation of the subscription for the server in the TestServer project.

nauful, 2024-07-02:
Hello,

Can you capture subscribing via uaexpert in Wireshark with no security and share that here? I would guess it subscribed to something besides (nodeid, value).

Christ0ph89, 2024-07-03:
Hi Nauful,

sure, I've capture the traffic from UAExpert and the LibUA client for the mentioned node, which is now ns=4;i=2801; after an structure update on the OPC server. The according dump is named after that at the end (..._libua / ..._uaexpert). 

What I've seen so far, the requests are nearly the same for the specific node, but in the LibUA client, the "NotifyDataChangeNotifications" method is not fired on an publish response for that node. Maybe there is some filter or switch case missing in LibUA?

[opc_nodeid_subscribe_libua.zip](https://github.com/user-attachments/files/16081289/opc_nodeid_subscribe_libua.zip)
[opc_nodeid_subscribe_uaexpert.zip](https://github.com/user-attachments/files/16081290/opc_nodeid_subscribe_uaexpert.zip)

nauful, 2024-07-03:
Hmm, it looks like that should have worked. Please check this line and step through to see if it reaches the call to `NotifyDataChangeNotifications`:
https://github.com/nauful/LibUA/blob/34253aa0c75cbd7e6988413c06fda7a5c35ca773/NET%20Core/LibUA/Client.cs#L4294

If decoding succeeds, it should call `NotifyDataChangeNotifications`. You override that in your client, for example:
https://github.com/nauful/LibUA/blob/34253aa0c75cbd7e6988413c06fda7a5c35ca773/NET%20Core/TestClient/Program.cs#L98

nauful, 2024-07-03:
Thanks, that gives me a hint. Let me investigate. I don't think I've seen an extension object in a variant before so this is a new case.

nauful, 2024-07-03:
Fixed, let me know if the latest version works.

Christ0ph89, 2024-07-03:
Looks good, great work 👍 . The only thing that seems to be missing is the source timestamp, the server timestamp is correct in the notification. Maybe another bug?

nauful, 2024-07-03:
There's no source timestamp in the data:
[image]

uaexpert requested both timestamps:
[image]

In your `CreateMonitoredItemsRequest`, request both source and server timestamps:
[image]

Christ0ph89, 2024-07-03:
Okay that was my fault, sorry for that 👍 .

Thanks again for the fast response and quick fix, everything is working now like expected :-)

closed 2024-07-03 by Christ0ph89