nauful/LibUA #179: Having trouble with Call method when method has Extension Object

Retx3D, 2024-07-26:
Hello nauful,  
first thank you for this very helpful library. 

I have been able to call a method successfully when this one only has classic variables in it but I am having trouble with the methods that uses ExtensionObject in parameters. 
the method I want to call is specified here : [https://reference.opcfoundation.org/MachineVision/v100/docs/7.5.2.4](https://reference.opcfoundation.org/MachineVision/v100/docs/7.5.2.4) 

here is my code to call the method : 
CallMethodRequest[] getRecipeListFilteredrequests = new CallMethodRequest[]
{
    new CallMethodRequest(new NodeId(2,5015), new NodeId(2,7018),new object[]{(string)"*", (string)"*",
        (Int32)2, (UInt32)0,
        (UInt32)0, (Int32)250})
};
LibUA.Core.CallMethodResult[] getRecipeListFilteredresults;
client.Call(getRecipeListFilteredrequests, out getRecipeListFilteredresults);

In the end I get a "BadInvalidArgument" error which I am pretty sure is linked to the fact that I am not using the extensionObject. But my tentatives to use them also did not work.

If you have any idea that would be awesome :) .

nauful, 2024-07-26:
Hello,

Could you capture the traffic in Wireshark with security disabled?

I guess you're writing a client and the server is rejecting the call. In this case, the extension object probably contains a serialized buffer which the server rejects.

bevanweiss, 2024-07-29:
I think the error message here is pretty spot on 'Bad... Invalid Argument'.
The method that you are calling `GetRecipeListFiltered`, wants some relatively complex typed arguments:
- [RecipeIdExternalDataType](https://reference.opcfoundation.org/MachineVision/v100/docs/12.8) externalId
- [ProductIdDataType](https://reference.opcfoundation.org/MachineVision/v100/docs/12.16) productId
- [TriStateBooleanDataType](https://reference.opcfoundation.org/MachineVision/v100/docs/12.3) isPrepared
- UInt32 maxResults
- UInt32 startIndex
- Int32 timeout

You only appear to be throwing strings in for the first two, and then an Int32 instead of the TriStateBooleanDataType.
I'm pretty sure that various data type designators would need to be included in the data sent across as the arguments, i.e. just sending a 'String' does not let the server interpret this as an appropriately typed RecipeIdExternalDataType.

Retx3D, 2024-08-02:
Hello All, 

sorry for the long answering time.
Thank you for your answers. 

@bevanweiss thanks for the input. I think I should then use the ExtensionObject class. I think I have an idea on how to work this out.

I will be able to work again on it in about a week and a half. 

@nauful : thanks for the WireShark suggestion. I did not even think of it...

closed 2024-08-02 by Retx3D

bevanweiss, 2024-08-02:
I don't think an ExtensionObject is applicable here.  Although I've never really delved into this part of OPC-UA.
My currently shallow understanding is that an ExtensionObject is more for 'unknown'/'changing' data shapes... i.e. if you accept a whole bunch of different parameter TYPES, then you can just accept an ExtensionObject with those types within it.  The Method definition would then be consistent regardless of what type is actually put into the ExtensionObject.  I haven't seen this used much in standard OPC-UA definitions.
The other use case I have seen is where the data itself is 'don't care'.. like it's passing through to something else... in which case an ExtensionObject would allow for it to be a bunch of different data shapes... but it's not the intermediates job to deal with them, it just pushes them through to the next layer.

For the methods that you were looking for, they are defined Structured DataTypes.  Which I believe are similar to the ValueTypes items 
https://github.com/nauful/LibUA/tree/master/NET%20Core/LibUA/ValueTypes

Which currently requires for them to be accommodated within the core of LibUA (there's not yet a concept of 'plugin' support for additional structured types).

I might be entirely wrong about this, and it might just work... Wireshark would give you a better idea if you try the 'simple' situation first... see what it looks like on the wire, and then you might need to revise.

nauful, 2024-08-03:
The easiest way to do this is create a `MemoryBuffer`, write your structure to that, then use that as the body for an `ExtensionObject`. Wireshark's opcua plugin can validate the structure you send according to the type specified for the `ExtensionObject`, which should show you any missing or misaligned fields.