nauful/LibUA #188: BadTooManyPublishRequests

Polyteres, 2024-09-20:
Hi,

We have an OPC server based on the sample server from this library and we are getting this message without any defined pattern (the application can be running without problems for hours). As workarounds we have increased the number of clients when we created the server to 400, the timeout to 40 and the MaxPublishRequests of the class NetDispatcherBase to 8192 but we still get this error. Is there any limitation in this regard?.

[image]

Best Regards, Borja Navarro.

nauful, 2024-09-20:
Hello,

It looks like the client is repeatedly sending a request for a publish but these requests are never satisfied. You should normally see:
1. Client sends a publish request.
2. Server writes new data.
3. Client receives a publish response to the request with the new data.
4. Client acknowledges 3 while sending another publish request.

It looks like this is happening:
1. Client sends a publish request.
2. None are used to send the client data.
3. Client does 1 again (which is incorrect behaviour).

If you can send me a wireshark capture with security disabled, I can take a look. Also pushed an update which fixes the count in the logger.

nauful, 2024-10-17:
Closed due to inactivity. Please reopen if you can capture this issue with Wiresharp.

closed 2024-10-17 by nauful