nauful/LibUA #189: Exception in background thread causing process to terminate

2sComplement, 2024-10-02:
Hello, and thanks for your work on this library.

As the title says, I'm running into an intermittent issue where a socket exception on a background thread is causing an application crash:

```
Unhandled exception. System.Net.Sockets.SocketException (10054): An existing connection was forcibly closed by the remote host.
   at System.Net.Sockets.Socket.UpdateStatusAfterSocketErrorAndThrowException(SocketError error, Boolean disconnectOnFailure, String callerName)
   at LibUA.Client.CloseConnection()
   at LibUA.Client.ThreadTarget()
```

It seems like the server is forcing the connection closed from its end during or before the call to `CloseConnection()`, causing this exception to be thrown. My problem is that I have no way of catching this exception.

What is the recommended way of handling this? Ideally I could catch this exception and recreate my client.

2sComplement, 2024-10-03:
Thanks! I will try it out and let you know in a day or so, since the issue was intermittent but not infrequent.

Do you think it would be worth having a way for consumers to optionally provide their own exception handlers for exceptions thrown in the background thread?

2sComplement, 2024-10-04:
I have not seen this issue since upgrading, so it seems like your fix is working. The exception is almost certainly being swallowed though which feels wrong, hence my question about providing an exception handler. Any thoughts on that? I'm happy to submit a PR if that helps!

nauful, 2024-10-04:
There's no straightforward way to collect async exceptions in .NET. You'll get either an error status code or an exception if you try to call any methods on a client with a closed connection now, and given that you can catch that synchronously, you can reopen the connection at that point.

closed 2024-10-04 by nauful