nauful/LibUA #190: If opc server not reachable, unhandled exception thrown, after long run, causing applicaiton exit abnormly, possibly due to thread competiting on socket.

xiaoyuvax, 2024-10-17:
```
private void ThreadTarget()
{
...
CloseConnection();   // -> this method call may cause Unhandled Thread Exception, possibly due to thead competiting on socket resources.
}
```
Exception: An existing connection was forcibly closed by the remote host.

Stacktrace (line no seems not match the source):
> /CurrentDomain_UnhandledException at offset 222 in file:line:column <filename unknown>:0:0
> /UpdateStatusAfterSocketErrorAndThrowException at offset 119 in file:line:column <filename unknown>:0:0
> /CloseConnection at offset 182 in file:line:column <filename unknown>:0:0
> /ThreadTarget at offset 323 in file:line:column <filename unknown>:0:0

xiaoyuvax, 2024-10-17:
possibly client.Connect()  is not threadsafe, as is what causes my problem.

bevanweiss, 2024-10-17:
What version of the library are you using? (and .NET Core / .NET Framework?)
This should be something like 1.0.2X

Ideally you'd also provide a minimal reproducible repository for this issue.

nauful, 2024-10-17:
This looks like a `SocketException` and should be fixed as of https://github.com/nauful/LibUA/commit/8051a828957317357b4a48b3c3b7de9fe28442fa in 1.0.28.

xiaoyuvax, 2024-10-18:
i used 1.0.24,
upgrade to 1.0.28, solved!
thx!

closed 2024-10-18 by xiaoyuvax