nauful/LibUA #191: Bad_TcpMessageTooLarge sending messages with more than 65535 bytes

schmelnik, 2024-11-05:
Sending larger messages from the server to a client (Matrikon OPC UA Explorer) is resulting with a Bad_TcpMessageTooLarge message on the client. The server itselfs sees just a BadIdentityTokenInvalid error as a result of it.

It looks like that this is a result of these two actions in NetDispatcher:

1. Method "DispatchHello"
Receives the "RecvBufferSize" and "SendBufferSize" from the client - limits these values 65535 bytes (the client supports larger sizes: 1MB), stores these limited values in the "config.TL.LocalConfig" structure and sends this structure to the client.
2. Method "DispatchMessage_SecureAndSend"
To calculate the number of chunks and the chunk size it is using "config.TL.RemoteConfig.RecvBufferSize" which is still indicating a larger size than communicated during the "Hello" operation. As the result messages larger than 65535 bytes are not splitted into multple chunks and the client is refusing to receive these large messages.

So I suggest to make the minimum buffer sizes configurable and use properly "config.TL.LocalConfig.RecvBufferSize" for the chunk calculation in "DispatchMessage_SecureAndSend" (probably there are more locations with similar issues)

nauful, 2024-11-05:
Try with the latest and let me know if that fixes your issue.

schmelnik, 2024-11-06:
Thanks for the quick repsonse.

I tried the described scenario with the latest build and it seems to resolve the described issues. To make sure that the problem is also fixed with multiple chunks (it turned out that all of my test data fits now into the increased buffer size) I decreased the chunk size again and verified that everything works fine even if the data has to be transmitted in multiple chunks.

Anyway some code remarks:
- I really like your change to get rid of the Local- and Remote-Config and replacing both of them by a TransportConfig
- You missed some changes in Net\LibUA\Client.cs - so actually the code does not compile
- The Client code now uses a different chunk size as the Server code. I would still recommend to make this value configurable like MaxMessageSize
- I guess MaxMessageSize and MaxChunkCount are related to send and receive operations. But by now MaxChunkCount is calculated from the size of the receive buffer. Since in theory the send and receive buffer size might be different, it should consider both sizes for calculating the MaxChunkCount value

nauful, 2024-11-06:
Fixed, should be good now. UA clients can request a chunk size from the server but reduce this appropriately to match server limits. 64k is the default limit for OPC UA, and everything larger should be broken into multiple chunks.

NetDispatcher will copy chunk and message limits from the client's request. It should ideally be the same size because, clients either have small memory limits (embedded) and are designed to work with low memory, or have large memory and have better performance with fewer chunks.

closed 2024-11-06 by nauful

schmelnik, 2024-11-06:
Any idea when the next release will be done?

nauful, 2024-11-06:
I'll push another nuget package after I have time to run against my test suite, probably within the week.

nauful, 2024-11-07:
I've published a new nuget package.