nauful/LibUA #192: CreateSubscription() receives data updates from simulated tags fluently, but not from actual PLC data updates, possible reason?

xiaoyuvax, 2024-11-06:
CreateSubscription() receives data updates from simulated tags fluently, but not from actual PLC data updates except for initial update immediately after subscription, no more data received, possible reason?

UaExpert monitors fluently too.

```
if (IsGood(CreateSubscription(requestedPublishingInterval, maxNotificationsPerPublish, true, priority, out uint subscrId)))
{
                SetPublishingMode(true, [subscrId], out uint[] respStatuses);

                var monitorRequests = tagsToMonitor.Select((i, n) => new MonitoredItemCreateRequest(i,
                    MonitoringMode.Reporting, new MonitoringParameters((uint)n, 0, null, 100, false))).ToArray();

                if (IsGood(CreateMonitoredItems(subscrId, TimestampsToReturn.Both, monitorRequests, out MonitoredItemCreateResult[] monitorCreateResults)))
                {...}
}
```

xiaoyuvax, 2024-11-07:
I made a test program exactly with the TestClient code from LibUA demos (the latest pull), in which:

```
client.CreateMonitoredItems(subscrId, TimestampsToReturn.Both,
                [
                    new MonitoredItemCreateRequest(
                        new ReadValueId(new NodeId(2, args[1]), NodeAttribute.Value, null, new QualifiedName()),  
                        MonitoringMode.Sampling,
                        new MonitoringParameters(clientHanlder, 500, null, 100, false)),
                ], out monitorCreateResults);
```

- above code monitors nothing, not even the initial updates, neither simulated nor real PLC tags.
- change to MonitoringMode.Reporting, above code works well with simulated tags but not for my PLC tags.
- while UaExpert always monitors ok.

nauful, 2024-11-09:
Can you share a Wireshark capture with security set to None with Wireshark, then with your libua client?

xiaoyuvax, 2024-11-11:
Sorry, not possible so far, it's found in a production environment. I don't even have similar environment to reproduce it locally, as is my headache too, otherwise, i would have located the line causing the problem in the source.
Finally, I had to reimplemented and done the client with h-opc-cli, and it works perfectly after i forked it and made minor changes.
If any further discover, i'll infom here.

nauful, 2024-11-11:
It's difficult to say what might be different because it could be anything from subscription configuration to connection/auth configuration.

closed 2024-11-11 by nauful