nauful/LibUA #197: Inconsistent "BadUnexpectedError" from OpenSecureChannel

LukeSawyers, 2024-11-29:
Before I start I just wanted to say thank you for making this library, this is a much more accessible alternative to the official repo.  This issue is the only thing currently giving me pause. 

`client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);`

Irregularly returns `BadUnexpectedError` when connecting to Omron NJ PLC embedded OPC UA Server. I say irregularly because it seems to depend on the configuration the code is run in - the issue almost never occurs in debug, sometimes in release, and quite often when running in docker. 

Any insight here would be helpful, I'm thinking this might be a race condition of some kind

nauful, 2024-11-29:
I can't reproduce this locally. Can you take a Wireshark capture of the connection attempt when it fails?

LukeSawyers, 2024-11-29:
[BadUnexpectedErrorCap.zip](https://github.com/user-attachments/files/17953767/BadUnexpectedErrorCap.zip)

Occurs 8 times before finally connecting

nauful, 2024-11-29:
It looks like the TCP connection is aborted on the client's side before a response is received. Can you try running `TestServer` and `TestClient` together?

LukeSawyers, 2024-11-29:
[TestClient&Server.zip](https://github.com/user-attachments/files/17953864/TestClient.Server.zip)

FYI Platform is Ubuntu 24.04, .NET 8.0.110

nauful, 2024-11-29:
I don't see any disconnects in this capture, did it work fine?

If so, it may be related to Ubuntu or the docker setup. The initial connection with security disabled is blocking, so I don't see anything that would cause a timing issue. The only error I can find is the TCP connection gets cut on the client side and the client doesn't receive any response from the server.

LukeSawyers, 2024-11-29:
I’m happy to dig into this further, would you be able to point out where this would be most likely to occur and any logging that could be done to track this down?

nauful, 2024-11-29:
`TestClient`: https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/TestClient/Program.cs#L137

This is the method in the UA client: https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L134

Incoming bytes are consumed here, perhaps check this for failures: https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L1382

If a message fails to process due to an error, this will trigger: https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L1207

If your environment closes the connection, this will trigger: https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L1127

nauful, 2024-12-25:
Closed as not reproducible. You can reopen if you find anything I can help with.

closed 2024-12-25 by nauful