nauful/LibUA #198: Status code = 0X80020000

nurgaleevarman, 2024-12-24:
I have a LibUA OPC server running.
UA Expert connects fine, but another client sometimes can't connect to the server. It gives an error **"Sent TL error 0x80020000"**
Security set to None.

OPC UA Server has 10k variables. 

How i can fix it?

nauful, 2024-12-24:
Can you please record a failed connection attempt with Wireshark and share the dump?

nurgaleevarman, 2024-12-25:
[tl error.pcapng.gz](https://github.com/user-attachments/files/18246208/tl.error.pcapng.gz)

line 3787, 7584, 11476

nauful, 2024-12-25:
[image]
Your read request is too large. Try reading 1k-10k nodes at a time. UAExpert reads in batches.

closed 2024-12-25 by nauful

nurgaleevarman, 2024-12-25:
Thank you!
Temporary solution is to increase the buffer.
`server = new LibUA.Server.Master(opcUA, port, 10, 30, 100, logger, **536870912**);`

nauful, 2024-12-26:
I would recommend using a smaller buffer as you will get unpredictable memory spikes from clients connecting, and potentially many performance issues.

nurgaleevarman, 2025-02-07:
Hello! This error is sent by the LibUA OPC server.
OPC UA Server uses over 10,000 tags with string identifiers.

nauful, 2025-02-12:
Use batches instead of very large single calls. This is unsupported behavior.