nauful/LibUA #199: LibUA monitor stops unreportedly in uncertain period of time without disconnection or exception, possible reason? while parellelly another OPC client monitors ok.

xiaoyuvax, 2025-01-14:
LibUA monitors only one tag simulated from Opc Server, it stops unreportedly in uncertain period of time (10+ mintues or up to half an hour or more) without disconnection or exception, nothing reported.

parellelly another opc client based on HDotOpc of the same parameters was run and it monitored on without break, everything goes normally.

possible reason?

nauful, 2025-01-14:
If I understand the issue, you're using the client to subscribe to a tag and get notifications back.

Please capture a wireshark for this event with security disabled, or debug this area:

When a data change notification comes in, this code is triggered https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L1566
Every data change notification leads to the client sending another publish request and the server will respond with the next notification async. The code above will set a flag to call publish request again:
https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L4383

xiaoyuvax, 2025-01-15:
so far, i guess might be certain exception thrown before this line that caused nextPublish remains false, thus leads to this notification/publish loop break:
 https://github.com/nauful/LibUA/blob/0794efda13076ad0bbf572a6dd2e001d39535f41/NET%20Core/LibUA/Client.cs#L1576

nauful, 2025-01-15:
I've moved nextPublish to a finally block if the data change dispatch throws an exception. Check if this solves your issue, and see if you get any exceptions from the notification handler or anywhere else.

xiaoyuvax, 2025-01-17:
After a 24hour test run, the problem hasn't shown, but myth is that despite i added a cacth between in order to print the exception messages, i never got it. The problem seems solved, but still needs extended test to see.  i'll report here if it occurs again.

it seems to be related with my previous issue:
https://github.com/nauful/LibUA/issues/192#issuecomment-2467969933

nauful, 2025-01-17:
Good, let me know if you see it again.

closed 2025-01-17 by nauful