nauful/LibUA #200: I can read a Node in the PLC

jariza-o, 2025-01-22:
Hello,

I like develop a SCADA program in C# .Net with the library, but is the first time that I develop in C# .Net and I can't read a Node. I show me code, I'm trying make a small program for I understand the most basic:
`using LibUA;
using LibUA.Core;

namespace Test_FREE
{
    public partial class Form1 : Form
    {
        public Form1()
        {
            InitializeComponent();
        }

        private void button1_Click(object sender, EventArgs e)
        {
            try
            {
                var client = new Client("192.168.1.100", 4840, 1000);
                var nodeId = new Node("ns=2;s=GVL.ST1_Q_CVY");
                var dataValue = client.Read(nodeId);
                if (dataValue != null)
                {
                    Console.WriteLine($"Valor del nodo {nodeId}: {dataValue.Value}");
                }
                else
                {
                    Console.WriteLine("No se pudo leer el valor del nodo.");
                }
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Error al leer el nodo: {ex.Message}");
            }
        }
    }
}`

Thanks!!

krosthw, 2025-01-23:
I suggest you to look at the [example](https://github.com/nauful/LibUA/blob/master/NET%20Core/TestClient/Program.cs) in the TestClient Folder.
You are missing some instructions before you can read.
**If your endpoint accepts anonymous authentication**, you can try this code:

     private void button1_Click(object sender, EventArgs e)
        {
            try
            {
                var client = new Client("192.168.1.100", 4840, 1000);
    			var messageSecurityMode = MessageSecurityMode.None;
                var securityPolicy = SecurityPolicy.None;
                var openRes = client .OpenSecureChannel(messageSecurityMode, securityPolicy, null);
                var createRes = client .CreateSession(appDesc, "urn:DemoApplication", 120);
                var activateRes = client .ActivateSession(new UserIdentityAnonymousToken("1"), new[] { "en" });
                var nodeId = new NodeId(2, "GVL.ST1_Q_CVY");
    			
    			var dataValue = client.Read(new ReadValueId[]
    				{
    			new ReadValueId(nodeId, NodeAttribute.Value, null, new QualifiedName(0, null)),
    
    				}, out DataValue[] dvs);
    			
                //then you look at dvs to see the result of the reading
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Error al leer el nodo: {ex.Message}");
            }
        }

jariza-o, 2025-01-23:
Hi,

Thansk for you reply.
I try this code, and when is declarated the var OpenRes, the program pass at the exception. I don't know the motive because the PLC is configured without security. For I can run the software I create the var appDesc, copying this var from you example code.

I attach the code and the PLC configuration.
Code:
`
using System.Security.AccessControl;
using LibUA;
using LibUA.Core;

namespace TestReplyGithub
{
    public partial class Form1 : Form
    {
        public Form1()
        {
            InitializeComponent();
        }

        private void button1_Click(object sender, EventArgs e)
        {
            try
            {
                var appDesc = new ApplicationDescription(
                "urn:DemoApplication", "uri:DemoApplication", new LocalizedText("UA SDK client"),
                ApplicationType.Client, null, null, null);

                var client = new Client("192.168.1.100", 4840, 1000);
                var messageSecurityMode = MessageSecurityMode.None;
                var securityPolicy = SecurityPolicy.None;
                var openRes = client.OpenSecureChannel(messageSecurityMode, securityPolicy, null);
                var createRes = client.CreateSession(appDesc, "urn:DemoApplication", 120);
                var activateRes = client.ActivateSession(new UserIdentityAnonymousToken("1"), new[] { "en" });
                var nodeId = new NodeId(2, "GVL.ST1_Q_CVY");

                var dataValue = client.Read(new ReadValueId[]
                {
            new ReadValueId(nodeId, NodeAttribute.Value, null, new QualifiedName(0, null)),

                    }, out DataValue[] dvs);

                //then you look at dvs to see the result of the reading
                MessageBox.Show(dvs[0].Value.ToString());
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Error al leer el nodo: {ex.Message}");
            }
        }
    }
}
`

PLC configuration:
[image]

nauful, 2025-01-23:
This isn't an issue specific to this library. If `OpenSecureChannel` returns an error status code, look at the status code to see what to do next. I would guess your server only accepts clients with specific certificates or client descriptions. You can look at `TestClient` to see how to set that up after adding those to your server.

closed 2025-01-23 by nauful