nauful/LibUA #202: ns=X;bs=xxxxx is not a valid nodeid.

thoj, 2025-02-02:
Hi

I noticed when testing against a server that expect opaque node ids that LibUA uses bs=0x in the ToString function on nodeId. This is not valid according to the spec. Only ns=x;b= is valid for ByteString/Opaque node ids: 

https://documentation.unified-automation.com/uasdkhp/1.4.1/html/_l2_ua_node_ids.html

https://github.com/nauful/LibUA/blob/18224cd68c8ba50fbca351551d546e25ae6c00dd/NET%20Core/LibUA/AddressSpace.cs#L126

I'm also would like TryParse to handle ns=x;b= But that is more of a feature. :)

thoj, 2025-02-02:
ns=2;guid=xxx is also wrong. Should be g= and with dashes: 

https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.12

nauful, 2025-02-02:
Thanks for catching that. I started adding this, but will accept your PR in favor of my commit: https://github.com/nauful/LibUA/commit/3b9099054c80dacc8094a2052d99c183c2686dfc

PR #203 (2025-02-02): Fix NodeId encoding formats and improve serialization
The standard only allows NodeId types of i, s, g and b. b is only allowed to be encoded as base64. Ref: https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.12#_Ref122646734
* Fix serialization in ToString() to follow standard. 
* Added support for b=0x and b=base64 in TryParse()
* Added support for g=GUID in TryParse()
* Added Tests for the Above.

Fixes #202
commit 25bb7d7c (2025-02-02): Fix NodeId encoding formats and improve serialization

- Fix/Add support for hex (`b=0x`), Base64 (`b=`), and GUID (`g=`) encoded NodeIds.
- Updated serialization to use Base64 for ByteString (`b=`) and proper GUID formatting (`g=`).
- Ensured compatibility with standard encoding schemes
  https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.12#_Ref122646734

Fixes #202
closed 2025-02-02 by nauful