nauful/LibUA #216: Adding Nodes Dynamically!

Noface07, 2025-04-30:
Hi, I've been working on a project where I bridge data from an OPC DA server (using TitaniumAS )  to OPC UA server (using LibUA). Everything works well so far — however, I noticed that the server currently returns BadServiceUnsupported when I try to Add nodes from the client (message type i=488), making it impossible to add nodes dynamically at runtime.

Is this a mistake on my side, or does the library not support adding the nodes dynamically through the client? Excuse my inexperience; I am a beginner.

Noface07, 2025-04-30:
I Implemented the functions in the server and I can now upload the nodes. Thank you for this wonderful package.

closed 2025-04-30 by Noface07

bevanweiss, 2025-05-05:
> I Implemented the functions in the server and I can now upload the nodes. Thank you for this wonderful package.

If some parts of this make sense to include into LibUA (i.e. a stub method in the Server base class or similar) it would be nice for you to submit your updates back to LibUA via a Pull Request.

nauful, 2025-05-05:
I'm glad this package has been useful for you. While it makes sense to add dynamic node creation to `TestServer` to show how it can be done, I wouldn't add this to the main library as servers have different rules around dynamic node creation. Some options include:
- Not allowed.
- Allowed for specific users based on authentication and the session object.
- Allowed for all users connected with security.
- Allowed only under a specific node in a hierarchy.
- Etc

In general, different business rules will enforce different logic for base servers. I would still appreciate any PRs for dynamic node creation to `TestServer` as a sample of how to do it for others.

mridah, 2025-05-18:
@Noface07 can you please share your code?