nauful/LibUA #226: Troubleshooting BadDecodingError on ActivateSessionRequest

flmarkus, 2025-07-30:
I'm building an OpcUA client with Workstation.UaClient.
For unit testing and mocking I have a LibUA server.

This is my simplified server setup:
```
    private readonly X509Certificate2 _serverCert;
    private readonly RSACryptoServiceProvider _serverCertKey;

    private readonly ApplicationDescription uaAppDesc;
    private readonly NodeObject TestNodeRoot;

    private readonly NodeVariable IsActiveNode;
    private readonly NodeVariable ProgramNameNode;

    private Master? _uaServer;

    public TestOpcUaServer(X509Certificate2 serverCert, RSACryptoServiceProvider serverCertKey)
    {
        _serverCert = serverCert;
        _serverCertKey = serverCertKey;

        uaAppDesc = new ApplicationDescription(
            "urn:OpcUaTestServer",
            "",
            new LocalizedText("en-US", "OPC UA Test Server"),
            ApplicationType.Server,
            null,
            null,
            null);

        var itemsRoot = new NodeObject(
            new NodeId(2, 0),
            new QualifiedName("Items"),
            new LocalizedText("Items"),
            new LocalizedText("Items"),
            0,
            0,
            0);

        // Objects organizes Items
        AddressSpaceTable[new NodeId(UAConst.ObjectsFolder)].References
            .Add(
                new ReferenceNode(
                    new NodeId(UAConst.Organizes),
                    new NodeId(2, 0), false));

        itemsRoot.References.Add(
            new ReferenceNode(
                new NodeId(UAConst.Organizes),
                new NodeId(UAConst.ObjectsFolder),
                true));

        AddressSpaceTable.TryAdd(itemsRoot.Id, itemsRoot);

        // Create CleaningSystem folder
        TestNodeRoot = new NodeObject(
            new NodeId(2, 1000),
            new QualifiedName("TestNode"),
            new LocalizedText("TestNode"),
            new LocalizedText("TestNode"),
            0,
            0,
            0);

        itemsRoot.References.Add(new ReferenceNode(new NodeId(UAConst.Organizes), TestNodeRoot.Id, false));
        TestNodeRoot.References.Add(new ReferenceNode(new NodeId(UAConst.Organizes), itemsRoot.Id, true));
        AddressSpaceTable.TryAdd(TestNodeRoot.Id, TestNodeRoot);

        var nodeTypeBool = new NodeId(0, 1);    // Boolean
        var nodeTypeString = new NodeId(0, 12); // String

        IsActiveNode = CreateVariable(1003, "IsActive", nodeTypeBool);
        ProgramNameNode = CreateVariable(1006, "ProgramName", nodeTypeString);

        // Init low level server with this class as application handling
        _uaServer = new Master(
            this,
            4840,
            10000,
            10,
            10,
            new LibUaLogWrapper(_logger));

        _uaServer.Start();
    }

    private NodeVariable CreateVariable(uint nodeId, string name, NodeId dataType)
    {
        var variable = new NodeVariable(
            new NodeId(2, nodeId),
            new QualifiedName(name),
            new LocalizedText(name),
            new LocalizedText(name),
            0,
            0,
            AccessLevel.CurrentRead,
            AccessLevel.CurrentRead,
            0,
            true,
            dataType);

        TestNodeRoot.References.Add(new ReferenceNode(new NodeId(UAConst.Organizes), variable.Id, false));
        variable.References.Add(new ReferenceNode(new NodeId(UAConst.Organizes), TestNodeRoot.Id, true));
        AddressSpaceTable.TryAdd(variable.Id, variable);

        return variable;
    }

    public override X509Certificate2 ApplicationCertificate => _serverCert;

    public override RSACryptoServiceProvider ApplicationPrivateKey => _serverCertKey;

    private ApplicationDescription CreateApplicationDescriptionFromEndpointHint(string endpointUrlHint)
    {
        string[] discoveryUrls = uaAppDesc.DiscoveryUrls;
        if (discoveryUrls == null && !string.IsNullOrEmpty(endpointUrlHint))
        {
            discoveryUrls = [endpointUrlHint];
        }

        return new ApplicationDescription(
            uaAppDesc.ApplicationUri,
            uaAppDesc.ProductUri,
            uaAppDesc.ApplicationName,
            uaAppDesc.Type,
            uaAppDesc.GatewayServerUri,
            uaAppDesc.DiscoveryProfileUri,
            discoveryUrls);
    }

    public override IList<EndpointDescription> GetEndpointDescriptions(string endpointUrlHint)
    {
        byte[] certStr = ApplicationCertificate.RawData;
        ApplicationDescription localAppDesc = CreateApplicationDescriptionFromEndpointHint(endpointUrlHint);

        var epNoSecurity = new EndpointDescription(
            endpointUrlHint, localAppDesc, certStr,
            MessageSecurityMode.None, LibUA.Core.Types.SLSecurityPolicyUris[(int)SecurityPolicy.None],
            [
                new UserTokenPolicy("0", UserTokenType.Anonymous, null, null, LibUA.Core.Types.SLSecurityPolicyUris[(int)SecurityPolicy.None]),
                new UserTokenPolicy("1", UserTokenType.UserName, null, null, LibUA.Core.Types.SLSecurityPolicyUris[(int)SecurityPolicy.None])
            ], LibUA.Core.Types.TransportProfileBinary, 0);

        var epSignBasic256 = new EndpointDescription(
            endpointUrlHint, localAppDesc, certStr,
            MessageSecurityMode.Sign, LibUA.Core.Types.SLSecurityPolicyUris[(int)SecurityPolicy.Basic256],
            [
                new UserTokenPolicy("0", UserTokenType.Anonymous, null, null, LibUA.Core.Types.SLSecurityPolicyUris[(int)SecurityPolicy.None]),
                new UserTokenPolicy("1", UserTokenType.UserName, null, null, LibUA.Core.Types.SLSecurityPolicyUris[(int)SecurityPolicy.Basic256Sha256])
            ], LibUA.Core.Types.TransportProfileBinary, 0);

        return
        [
            epNoSecurity //, epSignBasic256
        ];
    }
```

When the client connects I'm getting this error message after ActivateSessionRequest:
    Error: 0x80070000 [BadDecodingError]
    Reason: TL error 0x80070000

[image]

This is the request:

[image]

Does my client send an invalid request?
What exactly is wrong in the request?
It looks to me more like LibUA has issues parsing a correct request (maybe I'm wrong)
How can I debug the exact parsing issue?

Any help is greatly appreciated. Thanks a lot in advance.

nauful, 2025-08-03:
I don't see anything that stands out. Please try debugging the server through `DispatchMessage_ActivateSessionRequest`. If anything hits a `ErrorParseFail` case, it'll return a decoding error.

nauful, 2025-08-03:
I couldn't reproduce a decoding error by replaying the activate session request:
```csharp
private static int TestActivateSessionRequest()
{
    var packetBytes = new byte[]
    {
        0x01, 0x01, 0x02, 0x00, 0x83, 0x07, 0x56, 0x6b,
        0x2c, 0x01, 0xdc, 0x01, 0x03, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff,
        0x98, 0x3a, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0x01, 0x00, 0x00, 0x00, 0x02,
        0x00, 0x00, 0x00, 0x65, 0x6e, 0x01, 0x00, 0x41,
        0x01, 0x01, 0x05, 0x00, 0x00, 0x00, 0x01, 0x00,
        0x00, 0x00, 0x30, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff
    };

    var recvBuf = new MemoryBuffer(packetBytes);
    int ErrorParseFail = -1;

    if (!recvBuf.Decode(out RequestHeader reqHeader)) { return ErrorParseFail; }
    if (!recvBuf.DecodeUAString(out string clientSignatureAlgorithm)) { return ErrorParseFail; }
    if (!recvBuf.DecodeUAByteString(out byte[] clientSignature)) { return ErrorParseFail; }

    if (!recvBuf.Decode(out uint numSignedSoftwareCertificate)) { return ErrorParseFail; }
    if (numSignedSoftwareCertificate != 0xFFFFFFFFu)
    {
        for (uint i = 0; i < numSignedSoftwareCertificate; i++)
        {
            if (!recvBuf.DecodeUAString(out string _)) { return ErrorParseFail; }

            if (!recvBuf.DecodeUAByteString(out _)) { return ErrorParseFail; }
        }
    }

    if (!recvBuf.DecodeUAString(out string[] _)) { return ErrorParseFail; }

    if (!recvBuf.Decode(out NodeId userIdentityTokenTypeId)) { return ErrorParseFail; }
    if (!recvBuf.Decode(out byte userIdentityTokenBodyMask)) { return ErrorParseFail; }

    // Binary body
    if (userIdentityTokenBodyMask != 1)
    {
        return ErrorParseFail;
    }

    if (userIdentityTokenTypeId.NumericIdentifier == (uint)UserIdentityTokenType.Anonymous)
    {
        if (!recvBuf.Decode(out uint _)) { return ErrorParseFail; }
        if (!recvBuf.DecodeUAString(out string policyId)) { return ErrorParseFail; }
        Console.WriteLine("Anonymous token: {0}", policyId);
    }
    else if (userIdentityTokenTypeId.NumericIdentifier == (uint)UserIdentityTokenType.UserNameIdentityToken)
    {
        if (!recvBuf.Decode(out uint _)) { return ErrorParseFail; }
        if (!recvBuf.DecodeUAString(out string policyId)) { return ErrorParseFail; }
        if (!recvBuf.DecodeUAString(out string username)) { return ErrorParseFail; }
        if (!recvBuf.DecodeUAByteString(out byte[] password)) { return ErrorParseFail; }
        if (!recvBuf.DecodeUAString(out string algorithm)) { return ErrorParseFail; }

        if (string.IsNullOrEmpty(algorithm))
        {

        }
    }
    else
    {
        return ErrorParseFail;
    }

    if (!recvBuf.DecodeUAString(out string _)) { return ErrorParseFail; }

    if (!recvBuf.DecodeUAByteString(out _)) { return ErrorParseFail; }

    return 0;
}
```

flmarkus, 2025-08-04:
Thanks for pointing me to the right method for debugging.
While debugging I noticed, the decompiled source differs from latest in this repo.

Unfortunately I was targeting an old npm package. I was using https://www.nuget.org/packages/LibUA.
But this is obviously the latest one: https://www.nuget.org/packages/nauful-LibUA-core

Both packages point to this repo. Would it be possible to make it obvious that LibUA is deprecated and only nauful-LibUA-core should be used (IMHO I think the naming is a little bit misleading when comparing both names...)

This took me quite some time, only because of the outdated npm package.
Anyway I'm closing this issue.

nauful, 2025-08-04:
I don't own the original package. Someone else created it before I published to nuget.

flmarkus, 2025-08-05:
Ah, ok, I see. In this case it should be reported as abuse. It points to your github repo but does not belong to your project. Furthermore it contains code, that is quite similiar to your project's code (is your project a fork?).
This is critical in terms of security and stabillity.

nauful, 2025-08-05:
I've reported it to nuget and you may also do that. I originally wrote this library in 2015, and a user pushed it to nuget themselves before I started publishing to nuget.

flmarkus, 2025-08-05:
I've reported it as well.

nauful, 2025-08-11:
Fantastic, thanks!