nauful/LibUA #227: Why is data null when reading from opc?

HaonPAPA, 2025-08-21:
` var appDesc = new ApplicationDescription(
 "urn:KLAB_App", "uri:KLAB_App", new LocalizedText("OPCUA Client"),
 ApplicationType.Client, null, null, null);
 
 this._opcClient = new OpcClient(ip, port, 1000);

 // 서버 연결 뒤 서버 목록 조회 & 엔드 포인트 목록 조회
 this._opcClient.Connect();

 var code =  this._opcClient.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);
 this._opcClient.FindServers(out ApplicationDescription[] appDescs, new[] { "en" });
 this._opcClient.GetEndpoints(out EndpointDescription[] endpointDescs, new[] { "en" });
 this._opcClient.Disconnect();

 // 보안에 맞는 엔드포인트 선택
 var messageSecurityMode = MessageSecurityMode.SignAndEncrypt;
 var securityPolicy      = SecurityPolicy.Basic256Sha256;
 bool useAnonymousUser   = false;

 var endpointDesc = endpointDescs.First(e =>
e.SecurityMode == messageSecurityMode &&
e.SecurityPolicyUri == Types.SLSecurityPolicyUris[(int)securityPolicy]);
 byte[] serverCert = endpointDesc.ServerCertificate;

 // 서버에 연결 & 보안채널 & 세션 120초
 var connectRes = this._opcClient.Connect();
 var openRes = this._opcClient.OpenSecureChannel(messageSecurityMode, securityPolicy, serverCert);
 var createRes = this._opcClient.CreateSession(appDesc, "OPCUA Client", 120);

 StatusCode activateRes;
 if (useAnonymousUser)
 {
     // Will fail if this endpoint does not allow Anonymous user tokens
     string policyId = endpointDesc.UserIdentityTokens.First(e => e.TokenType == UserTokenType.Anonymous).PolicyId;
     activateRes = this._opcClient.ActivateSession(new UserIdentityAnonymousToken(policyId), new[] { "en" });
 }
 else
 {
     string policyId = endpointDesc.UserIdentityTokens.First(e => e.TokenType == UserTokenType.UserName).PolicyId;

     // 서버에서 endpointDesc.UserIdentityTokens 확인 후 PolicyId 가져오기
     string username = "root";
     string password = "root";

     // UserIdentityToken 생성
     var userToken = new UserIdentityUsernameToken(
         policyId,
         username,
         Encoding.UTF8.GetBytes(password),
         Types.SignatureAlgorithmRsaOaep
     );

     //// Will fail if this endpoint does not allow UserName user tokens
     //string policyId = endpointDesc.UserIdentityTokens.First(e => e.TokenType == UserTokenType.UserName).PolicyId;
     //activateRes = this._opcClient.ActivateSession(
     //    new UserIdentityUsernameToken(policyId, "plc-user",
     //        (new UTF8Encoding()).GetBytes("123"), Types.SignatureAlgorithmRsaOaep),
     //    new[] { "en" });

     // 세션 활성화
     var status = this._opcClient.ActivateSession(userToken, new[] { "en" });
 }

var nodeId = new NodeId(1, $"System.Temperature");
//var nodeId = new NodeId(2, 0);

var readRes = _opcClient.Read(new ReadValueId[]
{
    new ReadValueId(nodeId, NodeAttribute.Value, null, new QualifiedName(0, null))
}, out DataValue[] dvs);

if (dvs.Length == 0 || dvs.FirstOrDefault() == null)
    return null;

if (dvs.Length > 0 && dvs.FirstOrDefault().Value is double temp)
    return temp;
else
    return null;
`

I’m following the examples you provided to develop my application, but I’m having trouble reading values — the Read operation isn’t returning any data.

I’ve already verified the NodeId (both numeric and string forms) on the server. Could you explain why dvs is coming back as null?

Thanks in advance.

nauful, 2025-08-22:
Please connect to your server with security disabled and take a Wireshark capture of network traffic during the read. I don't see anything wrong in the code, so the server may return an error code, or the network traffic might show me another failure.

HaonPAPA, 2025-08-22:
The connection part has been resolved, thank you.
I have one more question: after connecting, if I do nothing and just wait for about 3–4 minutes, an error occurs at tcp.Client.Send(sendBuf.Buffer, sendBuf.Position, SocketFlags.None) inside the private StatusCode OpenSecureChannelInternal(bool renew) function.

The tcp object itself is null at that point. Internally, RenewSecureChannel() keeps being executed, right? Is this because the server isn’t sending me something? How can I resolve this issue?

nauful, 2025-08-22:
I would guess the server is expiring the secure channel faster than it should, which is unusual, or it disconnects with some other error, and I would need a wireshark capture to see that.

The secure channel's default lifetime is set here: https://github.com/nauful/LibUA/blob/e6697f9861e894805425e959b36b0221f9577154/NET-Core/LibUA/Client.cs#L196
300 seconds is usually fine for nearly all the servers I've tried. If this is too short, you can try lowering it to 30 seconds.

HaonPAPA, 2025-08-27:
> I would guess the server is expiring the secure channel faster than it should, which is unusual, or it disconnects with some other error, and I would need a wireshark capture to see that.
> 
> The secure channel's default lifetime is set here:
> 
> [LibUA/NET-Core/LibUA/Client.cs](https://github.com/nauful/LibUA/blob/e6697f9861e894805425e959b36b0221f9577154/NET-Core/LibUA/Client.cs#L196)
> 
> Line 196 in [e6697f9](/nauful/LibUA/commit/e6697f9861e894805425e959b36b0221f9577154)
> 
>  UInt32 reqLifetime = 30 * 10000; 
> 
> 300 seconds is usually fine for nearly all the servers I've tried. If this is too short, you can try lowering it to 30 seconds.

```
export const clientOptions: OPCUAClientOptions = {
  applicationName: 'R4',
  securityMode: MessageSecurityMode.SignAndEncrypt,
  securityPolicy: SecurityPolicy.Basic256Sha256,
  endpointMustExist: false,
  keepSessionAlive: true,
  connectionStrategy: {
    maxRetry: 3,
    initialDelay: 1000
  },
  requestedSessionTimeout: 3000,
  clientName: 'OPCUA Client'
};

/** 모니터링 아이템 구독 옵션 */
export const subscriptionOptions: ClientSubscriptionOptions = {
  requestedPublishingInterval: 100,
  requestedLifetimeCount: 100,
  requestedMaxKeepAliveCount: 10,
  maxNotificationsPerPublish: 100,
  publishingEnabled: true,
  priority: 10
};

```
These are options used in node-opcua. Does LibUA have similar options, or do they all need to be implemented manually?